goodrobots/maverick-web #3: add service workers

SamuelDudley, 2018-01-25:
example of cesium sw / pwa
https://github.com/Developer-Plexscape/cesium-pwa

SamuelDudley, 2018-02-16:
Investigate precache vs hand written service workers https://github.com/GoogleChromeLabs/sw-precache

cglusky, 2018-02-26:
@SamuelDudley I think this is the latest  https://github.com/googlechrome/workbox

fnoop, 2018-03-04:
Just FYI the web ui was started from this boilerplate:
https://github.com/vuetifyjs/pwa
It's a pretty good 'best practice' pwa setup with an application shell and service worker build out of the box.  The production build immediately works to a degree - if you turn your network off and reload the page instead of a dinosaur you get the maverick interface, but obviously with a bunch of errors in the console at this point.
Next step is to start trapping and reacting to network errors when they occur.

fnoop, 2018-05-08:
Service workers finally available in all major browsers - recently added to Safari and Edge:
https://jakearchibald.github.io/isserviceworkerready/

fnoop, 2018-05-08:
How to self-serve google fonts

fnoop, 2018-05-08:
MDN guides:
https://hacks.mozilla.org/2018/05/progressive-web-apps-core-guides-on-mdn-web-docs/

SamuelDudley, 2020-04-08:
Some notes on cache busting

https://medium.com/js-dojo/vuejs-pwa-cache-busting-8d09edd22a31