goodrobots/maverick-web #5: Website Authentication

fnoop, 2018-02-01:
Related:
https://github.com/goodrobots/maverick/issues/689

SamuelDudley, 2018-02-24:
What about supporting a white listed set of actions that a user can perform?

e.g. one user might not be able to modify the config / parameters but is able to plan / fly missions, wile another 'super user' can do all actions / modify everything. Just a thought.

fnoop, 2018-02-24:
The de facto standard would be LDAP for the authentication and RBAC for authorization.  I would have thought this would be done per API, rather than per web instance.  If there was auth done to the web instance, it would be for a user profile that might contain saved api logins.  But I think at the api level it we should probably look at standard RBAC - it probably fits in well with the typical use case and there will be lots of existing implmentations available to us.

cglusky, 2018-02-24:
Not sure how nginx will factor in to the new maverick but kong could be one option https://getkong.org/

fnoop, 2019-02-10:
Covered in #95

closed 2019-02-10 by fnoop