goodrobots/maverick-web #9: Logo

fnoop, 2018-02-06:
* themed or abstract?

SamuelDudley, 2018-03-04:
Is this only for maverick-web or maverick in general?

Irrespective my vote is with abstract. Does anyone know a good and cheap graphic designer or are willing to have a go at designing the logo?

fnoop, 2018-03-04:
I'm a terrible graphic designer but was going to have a go at this, at least for something temporary :)

fnoop, 2019-02-07:
@SamuelDudley Any idea where we should keep the artwork for the logos?  As in the illustrator files, svgs etc?  I can add them just to the maverick-web repo somewhere that isn't actually used, just as storage, or we could add them to a separate repo somewhere?

SamuelDudley, 2019-02-07:
We could have an asset repo at the goodrobots level and store them there. While I'm happy distributing the svg, jpg, etc perhaps the illustrator files should remain private?

fnoop, 2019-02-08:
Do you mean a private repo for the illustrator files? I had presumed we would just distribute everything in the spirit of openness, but I guess you're right - perhaps the illustrator files should remain private in case someone else uses them to alter the logo.  The QGC design guidelines are quite good, we could follow them loosely just as a general guideline:
https://qgroundcontrol.gitbooks.io/design-guide/content/

SamuelDudley, 2019-02-08:
Agreed, the QGC guidelines are nice. I just don't think we want to be publicly posting the illustrator files. The other formats, jpeg, PNG, whatever, go for it!

closed 2019-02-10 by fnoop