goodrobots/maverick-web #21: Suppress web-pack warnings in console (Cesium)

SamuelDudley, 2018-02-25:
related: https://github.com/webpack/webpack.js.org/issues/147
docs: https://webpack.js.org/configuration/stats/

Looks like we can set `warningsFilter` to suppress the warnings

```
[HMR] bundle has 3 warnings
webpack-internal:///896:184 ./node_modules/cesium/Source/Core/TaskProcessor.js
146:80-87 Critical dependency: require function is used in a way in which dependencies cannot be statically extracted
 @ ./node_modules/cesium/Source/Core/TaskProcessor.js
 @ ./node_modules/cesium/Source/Core/QuantizedMeshTerrainData.js
 @ ./node_modules/cesium/Source/Core/CesiumTerrainProvider.js
 @ ./node_modules/cesium/Source/Cesium.js
 @ ./node_modules/babel-loader/lib!./node_modules/vue-loader/lib/selector.js?type=script&index=0!./src/components/modules/planner/PlannerCesium.vue
 @ ./src/components/modules/planner/PlannerCesium.vue
 @ ./node_modules/babel-loader/lib!./node_modules/vue-loader/lib/selector.js?type=script&index=0!./src/components/modules/planner/PlannerModule.vue
 @ ./src/components/modules/planner/PlannerModule.vue
 @ ./src/router/index.js
 @ ./src/main.js
 @ multi ./build/dev-client ./src/main.js
./node_modules/cesium/Source/Core/buildModuleUrl.js
57:15-22 Critical dependency: require function is used in a way in which dependencies cannot be statically extracted
 @ ./node_modules/cesium/Source/Core/buildModuleUrl.js
 @ ./node_modules/cesium/Source/Cesium.js
 @ ./node_modules/babel-loader/lib!./node_modules/vue-loader/lib/selector.js?type=script&index=0!./src/components/modules/planner/PlannerCesium.vue
 @ ./src/components/modules/planner/PlannerCesium.vue
 @ ./node_modules/babel-loader/lib!./node_modules/vue-loader/lib/selector.js?type=script&index=0!./src/components/modules/planner/PlannerModule.vue
 @ ./src/components/modules/planner/PlannerModule.vue
 @ ./src/router/index.js
 @ ./src/main.js
 @ multi ./build/dev-client ./src/main.js
./node_modules/cesium/Source/Core/buildModuleUrl.js
80:77-84 Critical dependency: require function is used in a way in which dependencies cannot be statically extracted
 @ ./node_modules/cesium/Source/Core/buildModuleUrl.js
 @ ./node_modules/cesium/Source/Cesium.js
 @ ./node_modules/babel-loader/lib!./node_modules/vue-loader/lib/selector.js?type=script&index=0!./src/components/modules/planner/PlannerCesium.vue
 @ ./src/components/modules/planner/PlannerCesium.vue
 @ ./node_modules/babel-loader/lib!./node_modules/vue-loader/lib/selector.js?type=script&index=0!./src/components/modules/planner/PlannerModule.vue
 @ ./src/components/modules/planner/PlannerModule.vue
 @ ./src/router/index.js
 @ ./src/main.js
 @ multi ./build/dev-client ./src/main.js
```

fnoop, 2018-02-25:
Hmm, I've suppressed the build warnings OK but the runtime warnings just won't go away.
https://webpack.js.org/configuration/stats/
'This option does not have any effect when using the Node.js API.'
Just have to stay for now..

SamuelDudley, 2018-02-25:
Cool thanks for the removal of the build warnings! Closing.

closed 2018-03-31 by fnoop