goodrobots/maverick-web #22: AR/VR

fnoop, 2018-02-26:
A-frame: wrapper around webvr
https://aframe.io/
https://github.com/networked-aframe/networked-aframe
https://haydenlee.io/networked-aframe/

A-frame with Vue:
https://github.com/frederic-schwarz/aframe-vuejs-3dio
https://github.com/naidraikzir/vue-a-frame
http://blog.diepartments.de/webvr-webgl-vue-js-2-0-and-aframe-js-a-sweet-couple/

cglusky, 2018-02-26:
You may be interested in this project  https://github.com/AnalyticalGraphicsInc/gltf-pipeline

cglusky, 2018-02-26:
via @SamuelDudley http://gallinazo.flightgear.org/uas/autopilot-visualization-flight-track/

cglusky, 2018-02-26:
Just released Google AR SDK.  But Android only.  https://developers.google.com/ar/