goodrobots/maverick-web #24: Support Weather Data

cglusky, 2018-02-26:
Possibly with a radar overlay.

This is a slick little weather app that gets the point across:

https://play.google.com/store/apps/details?id=com.enzuredigital.weatherbomb&hl=en

fnoop, 2018-02-26:
Yeah absolutely.  When I want to go fly, the first thing I check is the weather and I keep watching it - particularly wind and precipitation.

cglusky, 2018-02-26:
Agree.  Just breaking out a few feature requests so we can keep an eye on The List a bit better.

If you do not have that android app I do recommend checking it out. Mostly for the swipe to manipulate time line idea.  I like it also because he has included pressure changes which is good for fishing.

brizey02, 2018-03-20:
I have nexrad tiles baked into mavcesium, getting windspeeds and other TMS sets shouldn't be a problem.

brizey02, 2018-03-21:
RADAR 5min updates
    layer = imageryLayers.addImageryProvider(new Cesium.WebMapServiceImageryProvider({ 
        url : '//mesonet.agron.iastate.edu/cgi-bin/wms/nexrad/n0r.cgi?', 
        layers : 'nexrad-n0r', 
        parameters : { 
            transparent : true, 
            format : 'image/png', 
            mydata : Math.random() 
        } 
    })); 
    layer.alpha = 0.5;