goodrobots/maverick-web #26: Support DTED

cglusky, 2018-02-26:
Digital Terrain Elevation Data.

May go by other names in other places.

Sourcing good data can be an issue.  Need to investigate latest.

fnoop, 2018-02-26:
I'm assuming this is taken care of inherently by the cesium 3d terrain sets?

cglusky, 2018-02-26:
Not sure.  There was some work done to ry to standardize things like buildings.  Lost track.

https://osmbuildings.org/

Might need to abstract the title of the issue once i get more info.

fnoop, 2018-02-26:
This is like organised squirreling.

cglusky, 2018-02-26:
yep

cglusky, 2018-02-26:
@fnoop yes, it should be part of the data set used by cesium as far as i can tell.  which i guess means we just need a way to query elevation data for whatever time and space you need.

https://www.citygml.org/

path planning and safety checks on min/max altitude etc would be the use case.

cglusky, 2018-02-26:
This could have a big impact on speed of adoption by removing some barriers.  it can be hard to generate good 3D maps.

https://cesium.com/blog/2017/01/24/cesium.com-update/

SamuelDudley, 2018-02-26:
@cglusky I have a cesium.com beta account. The new thing for rendering cities and other details is 3D tiles. http://www.aero3dpro.com.au/adelaide_3D_city_model_2017

brizey02, 2018-03-20:
Cesium has its own terrain data, You can also compile your own for your own use, its a bit of a pain though. The buildings and such should be other 3D tiles not related to terrain.