goodrobots/maverick-web #29: Support Closed Loop Survey Missions

cglusky, 2018-02-26:
Depends on #28 

Provide near real time feedback of camera capture during a mission.

Not looking for fully stitched ortho as that takes hours at this point.

Just thumbnails georectified so operator can ensure coverage during mission.

This will be payload and mission dependent so also a good chance to talk about how these might become GCS modules or some such.

SamuelDudley, 2018-03-02:
Related discussions
https://github.com/goodrobots/MAVCesium/issues/13

cglusky, 2018-03-03:
Interesting!  SO it seems stitching could be an option.  At least at lower res with some small time delay.  Alright, really need to get some dev frame stuff done so I can help test this stuff IRL later in year.

cglusky, 2018-03-03:
Hmm.  This could be a potential point to tie with with WebODM as well.  Also on The List.

SamuelDudley, 2018-12-21:
related workflow https://github.com/UASLab/ImageAnalysis