goodrobots/maverick-web #31: Add standard/advanced to parameter screens

fnoop, 2018-02-27:
Parameter screens should have clear dropdown to choose between standard and advanced parameters, with advanced hidden by default

SamuelDudley, 2018-03-03:
Note that PX4 does not categorise parameters in this way

fnoop, 2018-03-03:
In GQC parameter interface for PX4, there are 'Standard', 'Developer', 'Other' and 'System' areas.  Are these not obviously available from the parameter data?

SamuelDudley, 2018-03-03:
Not in the parameters that I generated from the code by running `make parameters_metadata` in `/srv/maverick/code/px4` unfortunately... perhaps there is another way to obtain this data from the codebase? I did note that the information that I have is in line with https://docs.px4.io/en/advanced_config/parameter_reference.html so maybe its a QGC managed list?

fnoop, 2018-03-03:
Yeah possibly.  It's really not that important, I'm actually wondering whether this should be implemented or not.  A parameter list/grouped interface such as we have currently is inherently 'advanced'.  'Standard' users should be using the other wizard/gui interfaces that will come, built on top of the underlying parameters.

fnoop, 2018-03-03:
Implementing this can cause confusion as if set to Standard by default a lot of parameters will be missing.  If we set it to Advanced by default, standard users will have to know to change it to standard.  And it involves an extra component to the UI, which if questionable usefulness is simply bad UX.
So I think maybe this is a bad idea and should be scrubbed?

SamuelDudley, 2018-03-03:
I personally like the idea of configuring the 'standard' parameters via a wizard and exposing all parameters how we currently are.

fnoop, 2018-03-03:
Cool.  case closed!

closed 2018-03-03 by fnoop