goodrobots/maverick-web #32: Parameter bitmasks better UX

fnoop, 2018-02-27:
Parameters that have a bitmask are currently displayed as dropdown, which is broken UX (eg arming mask).  Instead display a set of checkboxes to allow bitmask to be built.

SamuelDudley, 2018-03-03:
Additionally there are bitmask values which can have other meanings that are not captured by checkbox options. Similar to https://github.com/goodrobots/maverick-web/issues/33 an override should be available if the user wants to specify the integer bitmask.
An example is INS_LOG_BAT_MASK where the bitmask values are `{'2': 'IMU2', '1': 'IMU1', '4': 'IMU3'}` while the values are `{'1': 'First IMU', '0': 'None', '255': 'All'}`

fnoop, 2018-03-03:
Oh goody...

SamuelDudley, 2018-03-03:
I'm not sure what would happen if I set this param to 7... I expect that I would have the same behaviour as setting it to 255? 
On another note the default bitmask values are n^2. Would you prefer that the actual values are calculated in the API so that the web ui simple needs to add the decimal values together?

commit 4cb8c7e3 (2018-03-07): Lots of updates to params edit, layout fixes #32
closed 2018-03-07 by fnoop