goodrobots/maverick-web #37: Support 3D Data

cglusky, 2018-03-03:
This will be related to #28 

First, need to figure out how we should break out support of survey missions etc...

2D orthorectified imagery
3D point clouds and elevation data (lidar missions etc)
what else?

WebODM is using potree for 3D viewer
http://www.potree.org/

Appears to be some blocking issues with threejs and cesium integration.  Need more experts to look at this one.

Entwine is open source data back-end for point clouds; they appear to have cesium integration:
http://cesium.entwine.io/?resource=red-rocks

cglusky, 2018-03-03:
Will also need to decide on what level of support for 3D.  It will take serious compute resources to handle processing of 3D data in real time.  

Can it be done on TX1/2 class of machine on the robot or in the field?

https://github.com/AnalyticalGraphicsInc/3d-tiles

See example service from this tech at 

http://www.sitesee.io/

Given continued advancements in all converging tech likely that near instant generation of 3D assets will become norm/expectation.

This also leans toward VR/AR.

fnoop, 2018-03-03:
Supporting '3D Data' is like saying we're supporting 'drones' - that's too wide a term to discuss here.  And it depends on whether you mean generating, or processing, or displaying, as to the suitability for realtime and on what platforms.  This is an interesting topic and discussion, but suggest you raise separate issues for separate discussions of different '3d' areas.

cglusky, 2018-03-03:
Fixed and reopened.  Please note if I tag it with "research" it's very much a WIP.