goodrobots/maverick-web #38: Boolean parameters should be switch

fnoop, 2018-03-03:
Dropdown select for boolean parameters is poor UX.  Switch would be better.

fnoop, 2018-03-04:
Or boolean checkbox.

SamuelDudley, 2018-03-04:
Call me simple but I get confused by a single checkbox that is labelled `enable`. I'm not saying this is what you were considering but I thought it was worth mentioning.
It needs to be __super__ clear what the state is currently and what is about to be saved. Hope that makes sense!

fnoop, 2018-03-04:
I was thinking about a switch, like this:
https://vuetifyjs.com/en/components/selection-controls
I think they're nice and clear - they can turn a grey to indicate off or a green or other colour to indicate on, and can include text to that effect as well.  Nice ui feedback and great for touchscreens.

SamuelDudley, 2018-03-04:
Looks good! One thought... I would like to see `Enabled` and `Disabled` rather than `True` and `False` for our application

commit bdc9590f (2018-03-06): Update param edit with switch, slider, select, fixes #38
closed 2018-03-06 by fnoop