goodrobots/maverick-web #40: Save / Cancel buttons are obscured by keyboard on small devices

SamuelDudley, 2018-03-04:
Open to comments on this one, but it would be great if we didn't have to scroll down to save the parameters on small devices: 

[image]

SamuelDudley, 2018-03-04:
It may be possible to display the number only keyboard on mobile devices as well (but I think they are the same size).  Also seems the number keyboard has a range of issues: https://www.filamentgroup.com/lab/type-number.html

fnoop, 2018-03-06:
Moved save and close buttons up to title row in https://github.com/goodrobots/maverick-web/commit/bdc9590f90c9bbeb9306fc8934bce62d87230f8a.
Also un-flattened save button for better ux.

closed 2018-03-06 by fnoop