goodrobots/maverick-web #42: Prototype HUD screen

fnoop, 2018-03-08:
Good example of HUD screen:
https://discuss.ardupilot.org/t/new-mp-hud-i-have-been-working-on/26602

fnoop, 2018-03-08:
Artificial horizon is the obvious starting point, but should it be a fighter-like HUD, or a passenger-plane-like separate instrument?  Or option of both?

SamuelDudley, 2018-03-11:
Personally I like the fighter style HUD, but the importance should be readability and functionality over prettiness

fnoop, 2018-03-11:
These are awesome skeuomorphic dials
http://sebmatton.github.io/flightindicators/

fnoop, 2018-03-12:
https://github.com/eschnou/ardrone-webflight

SamuelDudley, 2018-03-12:
That ardrone-webflight software is interesting hits an lot of points we were discussing.

SamuelDudley, 2018-03-12:
https://youtu.be/JzRRk1U7fiA
What's nice is that the AP setpoint and current values can be clearly seen. RC stick inputs are a nice touch too.

fnoop, 2018-03-14:
https://github.com/richtr/Artificial-Horizon/blob/master/script/main.js

cglusky, 2018-03-31:
I would not worry about skeuomorphic dials.  I think it's safe to assume that is a learned interface for pilots and most sUAS operators will not have that background.

fnoop, 2018-03-31:
Yeah i've ditched the skeuomorphic stuff!  Just a HUD overlay for now

fnoop, 2018-04-01:
I've opted to put a lot of the info in dials into simple text indicators in the top bar.  This is a lot neater on the much smaller interfaces we use with drones (eg. tablet instead of a real cockpit..), and much easier to read for non-trained pilots (eg. me).