goodrobots/maverick-web #46: Config module very slow

fnoop, 2018-03-10:
Config module seems to have ground to a halt - very slow.  Is it computed values overloading?

fnoop, 2018-03-11:
Seems to be when a value is saved, there is a huge amount of processing going in the background to save the value and then process the received subscription update, in particular to merge the update with the stored graphql cache.

fnoop, 2018-03-11:
Delays occur in production build as well as dev.
Note it only seems to occur when talking/manipulating the dev SITL api params.

fnoop, 2018-03-11:
Confirmed it has something to do with dev SITL api.

fnoop, 2018-03-11:
Doing more testing on an old slow mobile device the pauses are apparent on all APIs, just much more pronounced on the dev SITL api.  I think it has to do with all the graphql parameter data being reactive and a non-keyed array of data.  Whenever the data comes in or changes, the frontend has to re-compute it into a keyed array so it the parameters can be accessed directly.
If it was possible to have the graphql data keyed, it would solve that problem.

SamuelDudley, 2018-03-11:
Can you give an example of how that would be done in the API?

fnoop, 2018-03-12:
I was hoping there was some way that graphql could serve the equivalent of a hash, but looking at the schema it can't do that, it has to be an array of hashes.  That's OK, I'll refactor the handling of the frontend code.

SamuelDudley, 2018-03-12:
Does the hash need to change when the data changes or be a unique hash that remains constant?

fnoop, 2018-03-12:
The hash/key really needs to be the id of the parameter, eg. ACCEL_Z_D, so we can easily access parameters - ie. params['ACCEL_Z_D'].  The alternatives are either to iterate through the array of hashes that come in through graphql and create a new keyed array (which is what is happening at the moment), or each time a value needs to be accessed the component has to scan through all params to access it.

SamuelDudley, 2018-03-12:
Understood. I'll see what is possible re the graphql return structure to simplify things for you.

fnoop, 2018-03-12:
The trouble at the moment is that every time the subscription returns an updated parameter, it forces a recompute of the entire secondary keyed array.  I think it's this that's causing the pauses/slowdowns.

fnoop, 2018-04-02:
parameter subscription updates status parameters frequently, such as STAT_RUNTIME and STAT_FLTTIME.  This forces a recompile of the parameter data internally.  Adding code to exclude STAT_ parameter updates from recomputations.

fnoop, 2018-04-02:
Every parameter update (mutation) seems to get at least two or three updates back through the subscription.

fnoop, 2018-04-02:
Look at gql pagination - no need to retrieve and process the entire set of parameters at a time.

fnoop, 2018-04-12:
Refactored parameter query and subscription logic to not need rekeying into keyed object, instead having an array lookup method, and also filters out STAT_ messages.  Interface and updates and subscription feedback is reasonably fast now on Chrome and Firefox, but inexplicably hangs Safari on both the desktop and mobile/tablet.

Worth noting that on traditional GCS there is a long pause everytime the GCS connects to a firmware to download parameters which can take a long time.  In our model the API is permanently connected to the firmware and preloads all the parameters and then watches them, so any client that connects to the API gets a very fast download of parameters (<1sec on a decent desktop).  This is actually a big plus point for our model.

fnoop, 2018-04-12:
Safari only hangs when the top bar is active and in the config module, must be some kind of race condition somewhere.  Separate issue.

fnoop, 2018-04-15:
It looks like when GCS such as qgroundcontrol and probably mission planner connects, it requests a stream of all parameters.  So instead of a single 'hit' of all parameters in a single query like the API provides to the graphql client, when qgroundcontrol requests the parameters one by one, this is also sent via the API to the graphql client, one by one.  
In addition, qgroundcontrol requests the streams at a much higher rate than the API is currently set to (1hz), and this then causes a very high rate of update to the graphql client which produces a very high CPU churn in the browser.  It is this that is probably the cause of hanging Safari, and also ramps up the CPU considerably in firefox and chrome.
In addition addition, using the vue-apollo library subscribeToMore mechanism (which adds a graphql subscription and updates a query by return) add significant cpu overhead to processing and causes blocking hangs in the browser, leading to unresponsiveness.  It turns out that using a more manual model of processing the graphql query and subscriptions provides a much more responsive browser experience.

fnoop, 2018-04-15:
So a couple of actions from the API end:
 - Stop the API sending a complete query/set of parameters after every subscription notice
 - Limit the number of subscription messages that the API can be sent out to a defineable rate - eg. a max of 2hz.  This is really the only way of not overloading the browser.

fnoop, 2018-04-15:
So interestingly the stateMessage seems to return at 1Hz and provides a full query and subscription every second.
vfrHudMessage is being provided at very high rate (5-10hz+), and does NOT provide a full query, just a subscription update.

fnoop, 2018-04-15:
Changing graphql client code to manual query+subscription model, like params component, has helped significantly.  So the automatic vue-apollo subscribeToMore definitely has performance problems.

fnoop, 2018-04-15:
Safari still hangs the UI but has less cpu usage and still does some updates, and if the *display* of the topnav data is turned off then safari is OK and responsive.  The subscription data is still processed at fast rate (10hz+) so it's the display of the data at that rate that Safari can't cope with.

fnoop, 2018-04-15:
Add a ticker mechanism to the top navbar which limits the frequency that the subscription updates can update the data containers, which in turns limits the visual component updates.  Set at 2hz, this now makes Safari useable if still laggy, and chrome quite useable.  In the planner module, it limits the cpu churn to very low amounts, in the config module the cpu usage is still quite high although useable.  This is certainly due to the background sending of the entire parameter set every subscription, which happens regularly due to STAT_, COMPASS_ etc updates.
This must be turned off at the API, nothing more we can do about this in the client.
However the ticker mechanism at least limits the damage that the API can do even if it's set at a high rate.  As the client has no control over the API, this is a good safety to build in anyway.

fnoop, 2018-04-15:
Added ticker mechanism to cockpit hud, limiting to 2hz drops cpu pegging down to 50% from 100%+

cglusky, 2018-04-15:
you think something like this would help?

https://github.com/convoyinc/apollo-cache-hermes

fnoop, 2018-04-15:
@cglusky v.interesting, afraid it's above my head how to integrate that or even diagnose where the actual bottleneck is in the low level front-end logic.  I'm just trying to get it not to crash at the moment for the proof of concept, we can look at optimising in the future.  Will keep this in mind!

cglusky, 2018-04-15:
understood.  was most intrigued by the write up in the wiki as to why they thought they needed it.  seemed to be along the lines of what you are experiencing re performance.

fnoop, 2018-04-16:
Refactored code now works much better even in Safari.  Config filter screen still lags a bit in Safari as the entire parameter set is pushed from the API and processed in the background constantly.  This still needs to be fixed at the API end:
https://github.com/goodrobots/maverick-api/issues/34

closed 2018-04-16 by fnoop