goodrobots/maverick-web #95: Add support for -api authentication

fnoop, 2018-12-31:
Initially use http header authorization tokens, eventually add support for session based auth with graphql mutations.

fnoop, 2019-01-02:
onLogin calls `await apolloClient.resetStore()` which causes `Error Network error: Store reset while query was in flight(not completed in link chain)`.
In addition ` if (apolloClient.wsClient) restartWebsockets(apolloClient.wsClient)` causes a websocket error: `WebSocket connection to 'ws://dev.maverick.one:7000/subscriptions' failed: WebSocket is closed before the connection is established.`

A cleaner way would be to add a 'logged in' state to each client and set query.skip property to false by default for those queries that need login, and only set to true when client is logged in.

SamuelDudley, 2019-01-02:
What would be required in -api to support this? Do you need a published authentication status?

fnoop, 2019-01-02:
I think it can be all done in the client-side

fnoop, 2019-01-12:
For now onLogin sets vuex state: `state.apis[data.api].auth`