goodrobots/maverick-web #136: Update to vuetify 2.x

fnoop, 2019-08-01:
Migrate from stylus to sass

fnoop, 2019-08-01:
https://vuetifyjs.com/en/customization/sass-variables

fnoop, 2019-08-01:
Major changes to vuetify2.  Default install is now alacarted with treeshaking, and webpack 4.
https://vuetifyjs.com/en/customization/a-la-carte

Start (another) new project and import the old one.

fnoop, 2019-08-05:
https://github.com/vuetifyjs/vuetify/releases#user-content-upgrade-guide

fnoop, 2019-08-05:
https://github.com/vuetifyjs/vuetify/releases/tag/v2.0.0

fnoop, 2019-08-05:
Need to go through and convert all slot activators:
```
Regular activator
<!-- v1.5 -->

<v-dialog>
  <v-btn slot="activator">...</v-btn>
</v-dialog>
<!-- v2.0 -->

<v-dialog>
  <template v-slot:activator="{ on }"
    <v-btn v-on="on">...</v-btn>
  </template>
</v-dialog>
```

fnoop, 2019-08-05:
Need to convert stylus mods:
```
// Import custom styles to override Vuetify defaults
// import '../stylus/main.styl'

// import 'vuetify/src/stylus/app.styl'
import colors from 'vuetify/es5/util/colors'
```

fnoop, 2019-08-05:
Fix FAB

fnoop, 2019-09-22:
Reopening, need to fix FAB

fnoop, 2019-09-23:
https://vuetifyjs.com/en/components/floating-action-buttons#fab-with-speed-dial

SamuelDudley, 2020-02-02:
@fnoop is this good to close?

fnoop, 2020-02-02:
Yes!

closed 2020-02-02 by fnoop