goodrobots/maverick-web #142: Add initial maverick configuration ui

SamuelDudley, 2020-01-29:
- [ ] interface for maverick commands, start, stop, etc...
- [ ] interface to view service status
- [ ] interface to view and edit config files

SamuelDudley, 2020-01-29:
How vscode handles configuration from a form perspective (which in the background is a json file)
[image]

SamuelDudley, 2020-02-02:
Basic interface to view and edit configuration files is complete. Working on a UI for commands and system processes now.

SamuelDudley, 2020-02-25:
https://github.com/xtermjs/xterm.js
Use terminal for maverick command feedback

fnoop, 2020-02-25:
ooh xterm.js is a nice idea!

SamuelDudley, 2020-02-26:
added the -web component of xterm.js https://github.com/goodrobots/maverick-web/commit/dc27bb11fe104abbde747e6f53bb9de513c84d5c
just hacked it in, but we can link this to -api for two outcomes:

1. Show realtime terminal output from the process runner (e.g. `maverick configure`)
https://github.com/goodrobots/maverick-api/blob/7ef29f4283e618320710c825c3b21513ab62bb96/maverick_api/modules/base/util/process_runner.py

2. Allow the user to interact with the terminal from the webui using the process runner

SamuelDudley, 2020-02-26:
https://github.com/wavesoft/local-echo/blob/master/lib/LocalEchoController.js

fnoop, 2020-02-26:
awesome!  would be great to have a mav terminal built in

SamuelDudley, 2020-03-17:
The terminal support was working (ish) prior to the API refactor in -web. It would be nice to get that back as it saves a user from needing to SSH in for basic stuff.

fnoop, 2020-03-28:
'Commands' re-instated.  If it's going to be a full terminal (which is great), should probably be renamed to Console/Terminal.
Console works-ish, restricted to a particular directory and the output is a bit hoky.  If we can get it working better will be fantastic to have.

SamuelDudley, 2020-03-28:
Yeah, the plan is to have the full terminal working. -api is okay at this stage, the remaining work is in the -web component.