goodrobots/maverick-web #150: Listen for -api broadcasts

fnoop, 2020-02-29:
-web should listen for -api discovery broadcasts, to find new -api instances.

fnoop, 2020-03-05:
Looks like this is the most up to date and popular mdns library:
https://github.com/agnat/node_mdns

fnoop, 2020-03-05:
@SamuelDudley Bad news :(.  There are lots of zeroconf/mdns libraries for javascript/node.js, but none of them work in the browser.  Browsers are fundamentally incapable of talking udp, which is what mdns runs over.  So for now, I don't think we're going to get discovery from the browser.

closed 2020-03-05 by fnoop

SamuelDudley, 2020-03-06:
Darn... This brings me back to the discussion we had on gitter.

What if we had a separate (new) service called `maverick-find` or something that is ultimately a super trimmed down maverick-api with only one job: identify maverick services and systems via zeroconf and advertise them all at a single well known endpoint on a maverick system.

e.g. we pick a easy to remember port and this service provides two things:

1. a web interface of links / network addresses that a user can click to interact with
2. programmatic access to this information via a REST and / or graphql interface

This would then enable -web to query this know endpoint and retrieve a list of available -apis (and other maverick systems)

fnoop, 2020-03-06:
Why not just add a module to -api that listens for other -api zeroconf broadcasts, collates them and then broadcasts to connected -web clients through graphql messages?  I guess that's what you're suggesting, just doing it in a module of the existing -api system seems simpler than creating a new micro-service to do the job?

SamuelDudley, 2020-03-06:
So we could do that... But then we still there the issue of not knowing where the -apis are until we connect to one. Having it at a predetermined port irrespective of if they are connected to sitl or fc would be an improvement.

fnoop, 2020-03-06:
That still doesn't solve the actual discovery bit - finding the location (hostname/ip address) of the service, not sure it would really help enough to be worth the extra effort?

SamuelDudley, 2020-03-06:
I'm convinced. I'm letting it go ;)

fnoop, 2020-03-06:
What we can do is go back to creating a default api if none already exist, using the hostname of the website that the user is already connecting to (so we know the hostname is valid).  It can automatically define and connect to that api, and then that api can subsequently send any nearby -api that it detects from zeroconf through a graphql subscription.  That would be pretty automatic - actually pretty cool.

SamuelDudley, 2020-03-06:
Is there some port forwarding magic we could do to have at least one API instance at a predictable port, like a load balancer of sorts? That way we can hit the same location for discovery?

SamuelDudley, 2020-03-06:
We could seperate out the discovery graphql endpoint and load balance that across available -apis leave the other stuff as it is... Bad idea?

fnoop, 2020-03-06:
https://www.w3.org/TR/discovery-api/
discontinued :(
I'm amazed there's absolutely no discovery ability in browsers at all..

fnoop, 2020-03-06:
OK yeah that's not a bad idea actually.  Actually your idea of a microservice on a reserved port is a very good idea.  There are a ton of services running on a ton of ports on a typical maverick install (and all the ports are configurable so not always the same), and having discovery service on a known port for all these services is a very good idea.

SamuelDudley, 2020-03-06:
Hey, I didn't call it a microservice ;) I'm easy, the port forwarding load balancer for discovery and the microservice will be equal work.