goodrobots/maverick-web #161: To history or not to history, that is the question?

fnoop, 2020-03-18:
If we want to set publicPath to '' or './' in vue.config.js, to allow relative pathing, then we can't use vue-router history mode:

https://cli.vuejs.org/config/#publicpath
https://router.vuejs.org/guide/essentials/history-mode.html

The difference is that we get 'prettier' URLs.  The difference is:
 http://maverick-ubuntuvm.local/web/maverick/video
vs
 http://maverick-ubuntuvm.local/web/maverick/#/video

fnoop, 2020-03-18:
This allows us to have the -dist version running on any URL (eg. / or /web/maverick, or /maverick-web as in the github pages), so it's quite useful.

SamuelDudley, 2020-03-19:
> This allows us to have the -dist version running on any URL (eg. / or /web/maverick, or /maverick-web as in the github pages), so it's quite useful.

I think the utility of this far outweighs the hashes in the urls

closed 2020-03-19 by fnoop