goodrobots/maverick-web #172: webrtc stream fails

SamuelDudley, 2020-03-30:
stream fails to start with the following error in the dev console
[image]

note: `webrtc` service on the pi is being spammed with keep alive requests (tens of requests per second) not sure if these are related

SamuelDudley, 2020-03-30:
restarting the `webrtc` service and refreshing the page, the stream connects and works.
`maverick log webrtc` shows lots and lots and lots of keep alive requests while the stream is playing and is visible. When the stream is active and the browser tab is switched so the video is in a background tab the keep alive rate drops to 1Hz.

fnoop, 2020-03-30:
Yes noticed this too, meant to add an issue in maverick project - it's a startup ordering issue that needs dependencies added to the systemd manifests I think.

closed 2020-04-01 by SamuelDudley