goodrobots/maverick-web #174: video stream browser compatibility tracking

SamuelDudley, 2020-04-01:
Starting this issue to track known working / not working configurations
- [x] chrome + linux `Working ok`
- [ ] firefox + linux `Shows buffering but stream does not start.` The following is in the console:
[image]
- [ ] chrome + ios (ipad) `Not working. Hard to say if this is a SSL issue or a webrtc issue`
debugging ios is a pain... https://9to5google.com/2019/03/25/chrome-ios-debugging-tool/ 
- [ ] safari + ios (ipad) `Not working. Hard to say if this is a SSL issue or a webrtc issue`
- [ ] chrome + android `unable to access pi hostname to verify` 
- [ ] chrome + windows 10 `was working when last tested, however generally unable to access pi hostname to verify`

fnoop, 2020-04-01:
Thanks this is needed!  I'll add MacOS options.

fnoop, 2020-04-01:
Doesn't work in safari on macos.  No warnings or errors.  Safari sucks.